gpstar81/GPStar-proton-pack #423: [Bug]: Neutrona Wand Volume Doesn't Adjust Independently.

The-Y33TER, 2025-02-26:
### What actions were attempted?

I followed the instructions on page 6 of the manual, pack switch off, wand all switches off, hold intensify and adjust via top knob.

### What actually happened?

The pack volume adjustment was altered as well. The wand *did* get to the level I wanted, but the pack was obnoxiously loud.

### What was expected?

The wand's independent adjustment should adjust the wand's volume independently.

### Firmware Version

5.4.3

### Mode in Use

Super Hero / Video Game

### Notable Hardware

Life size spirit, modded Spengler V1. Supreme kit, 28Seg, Hat Lights, RGB Vent/Arcolectric, 50LED barrel, 15LED Powercell, 9LED Cyclotron.

### Homework Completed

- [x] I did my part!

The-Y33TER, 2025-02-28:
> Converted this to a feature request because it is not in any way a bug (you were adjusting _system_ volume via the wand, not the wand volume, see [HERE](https://github.com/gpstar81/GPStar-proton-pack/blob/main/OPERATION.md#quick-audio-volume-adjustment)). However, rather than just close this issue as not a bug, I will note that others have requested the ability to adjust wand volume independent of the pack (especially, in specific, the default startup volume) and yet there was no open feature request for this.

Ohhhh. Yeah, makes sense. The verbiage used was a little confusing to me. The wording used for the mode original section says the wand volume though, so is this a mode by mode thing?

nomakewan, 2025-02-28:
> Ohhhh. Yeah, makes sense. The verbiage used was a little confusing to me. The wording used for the mode original section says the wand volume though, so is this a mode by mode thing?

Nope, just a typo. I'll go ahead and fix typo that as part of the 5.4.4 update. ;)

Right now the huge focus is on getting GPStar Lite up to where I want to see it. Once that's done the 5.x branch is pretty much tapped out for features, I think. We've wanted to jump to 6.x development for months but I insisted on making sure 5.x was stable and feature-complete (especially for Lite customers) before that.

The-Y33TER, 2025-03-02:
> Nope, just a typo. I'll go ahead and fix typo that as part of the 5.4.4 update. ;)
>
> Right now the huge focus is on getting GPStar Lite up to where I want to see it. Once that's done the 5.x branch [...]

hehehehehe sick (i am so excited, but I need to order SD card extenders)

nomakewan, 2025-04-22:
Hey there!

Wanted to follow up on this a bit. I've got this done for 5.4.4 (it's a hotly-requested item now especially with how the stock GPStar Amplifiers work), and in doing so I had a look at both the existing documentation and the code in 5.4.3. The documentation (as of 5.4.3) was correct that the quick volume adjustment was system volume only, and so _technically_ I was correct that this was not a bug. However, in reality there was still legacy code sitting inside the wand that was trying to do wand-only adjustments (but never could because that part of the code was unreachable). So in fact this was both a bug and a feature! Super weird!

The code is now cleaned up. Adjusting pack volume will now only adjust pack volume, and the quick adjustment on the wand can now independently adjust pack and wand volume (as well as music volume). Additionally, by default the wand will always start at 100% volume, but can be lowered using the quick adjust should someone feel the need.

closed 2025-04-22 by nomakewan