gpstar81/GPStar-proton-pack #425: [Feature]: Less violent vibration on the wand?

The-Y33TER, 2025-03-02:
### What would you like to add in terms of software changes?

Sorry for making so many issues, but this one bugs me. When I have the wand vibration on, it’s… SUPER loud. Like, obnoxiously loud. I turned it off, because it overpowers basically ALL of the sound effects, even the pack at a moderate volume level. I have a gen 1 spengler, does that make the vibration motor take less voltage to get loud, like a 3.3v motor instead of 5? Or is it just insanely loud 100% of the time with the wand vibration on? If so, it would be great if 6.0.0 permitted the user to make the vibration on the wand about the same as it was stock (maybe a little more) I’m the type of person to want the idle vibration on, but the intensity just… ruins it. Any possibility to add more options, like a “hold activate/barrel wing and use intensity knob to adjust” style option? 

### Would this request involve any specific hardware?

Nothing.

### Homework Completed

- [x] I did my part!

nomakewan, 2025-03-02:
I can reduce the idle vibration on the wand, sure. It's literally just a single PWM value. And I just so happen to have a stock wand handy that I've been using to tune the GPStar Lite with.

I'll see if I can't just get that adjustment done for 5.4.4.

If you set the vibration mode to "vibrate only while firing," is the firing vibration level fine? We made it rather intense on purpose since, yanno, firing.

The-Y33TER, 2025-03-03:
I'll check when I'm home, I can't remember. Trying to figure out a way to mount the pouch for the manual to my mobo so I can have it at cons lol (physical media > digital copy)

nomakewan, 2025-04-22:
I've reduced the base wand vibration value to match the base pack vibration values (base vibration is now 6% instead of 25.5%). This is included as part of 5.4.4.

closed 2025-04-22 by nomakewan