gpstar81/GPStar-proton-pack #426: [Support]: Music File Troubles.

The-Y33TER, 2025-03-11:
### What do you need assistance with?

I'm having issues with playing music. I've got all my music files as 500 onward, .wav files & no metadata. Still no dice. What's going on?

### Firmware Version

5.4.3

### Homework Completed

- [x] I did my part!

nomakewan, 2025-03-12:
We need more information about your music files first.

Please use a tool such as [MediaInfo](https://mediaarea.net/en/MediaInfo/Download) to output text-based specifications for your music files. It should look like this in MediaFire's text output mode:

```
General
Complete name                            : S:\gbsounds\500_library_mono.wav
Format                                   : Wave
Format settings                          : PcmWaveformat
File size                                : 34.4 MiB
Duration                                 : 3 min 24 s
Overall bit rate mode                    : Constant
Overall bit rate                         : 1 411 kb/s

Audio
Format                                   : PCM
Format settings                          : Little / Signed
Codec ID                                 : 1
Duration                                 : 3 min 24 s
Bit rate mode                            : Constant
Bit rate                                 : 1 411.2 kb/s
Channel(s)                               : 2 channels
Sampling rate                            : 44.1 kHz
Bit depth                                : 16 bits
Stream size                              : 34.4 MiB (100%)
```

Once you provide some examples of your music files we can proceed.

The-Y33TER, 2025-03-19:
### Example of the piano jingle I use on every pack I build to indicate music mode has started. 
```
General
Complete name                            : C:\Users\w\Downloads\PackMusic\500_Jingle.wav
Format                                   : Wave
Format settings                          : PcmWaveformat
File size                                : 1.31 MiB
Duration                                 : 7 s 175 ms
Overall bit rate mode                    : Constant
Overall bit rate                         : 1 536 kb/s

Audio
Format                                   : PCM
Format settings                          : Little / Signed
Codec ID                                 : 1
Duration                                 : 7 s 175 ms
Bit rate mode                            : Constant
Bit rate                                 : 1 536 kb/s
Channel(s)                               : 2 channels
Sampling rate                            : 48.0 kHz
Bit depth                                : 16 bits
Stream size                              : 1.31 MiB (100%)

```
### Example 2: The Elmer Bernstein Theme, the first music track I want to play (after the jingle)
```
General
Complete name                            : C:\Users\w\Downloads\PackMusic\501_Bernstein.wav
Format                                   : Wave
Format settings                          : PcmWaveformat
File size                                : 33.2 MiB
Duration                                 : 3 min 1 s
Overall bit rate mode                    : Constant
Overall bit rate                         : 1 536 kb/s
Track name                               : Ghostbusters Soundtrack (1984) 8 - Main Title Theme (Elmer Bernstein)
Director                                 : Mr. Marshmello
Writing application                      : Lavf60.3.100
Conformance errors                       : 2
 RIFF                                    : Yes
  General compliance                     : File size 34776616 is less than expected size 4294967303 (offset 0x8)
 WAVE                                    : Yes
  General compliance                     : Element size 4294967291 is more than maximal permitted size 34776604 (offset 0xC)

Audio
Format                                   : PCM
Format settings                          : Little / Signed
Codec ID                                 : 1
Duration                                 : 3 min 1 s
Bit rate mode                            : Constant
Bit rate                                 : 1 536 kb/s
Channel(s)                               : 2 channels
Sampling rate                            : 48.0 kHz
Bit depth                                : 16 bits
Stream size                              : 33.2 MiB (100%)
```

nomakewan, 2025-03-19:
Also, you still have metadata in both WAV files. All metadata should be stripped from the files.

The-Y33TER, 2025-03-26:
> Both of those files are 48.0 kHz, as you can see clearly in the text output for "Sampling rate".
> 
> Both the WAV Trigger and GPStar Audio require 44.1 kHz WAV files.

I don't know how to fix this. What should I do?

The-Y33TER, 2025-03-26:
> Also, you still have metadata in the second WAV file. All metadata should be stripped from all WAV files.

I don't know how to remove it. I tried a bunch of stuff, but no dice.

gpstar81, 2025-03-26:
> > Also, you still have metadata in the second WAV file. All metadata should be stripped from all WAV files.
> 
> I don't know how to remove it. I tried a bunch of stuff, but no dice.

With Audacity, when you export to a WAV file, there is a edit metadata button, then you can clear it all. You can also use Audacity to resample your WAV files to 44.1kHz by going to TRACKS -> RESAMPLE

nomakewan, 2025-03-26:
The method gpstar81 posted works perfectly. Can recommend.

You can also use the tool `mp3tag` for Windows to remove metadata from WAV files (I in fact use it to make sure the Github files don't have metadata in them).

The-Y33TER, 2025-03-30:
> The method gpstar81 posted works perfectly. Can recommend.
> 
> You can also use the tool `mp3tag` for Windows to remove metadata from WAV files (I in fact use it to make sure the Github files don't have metadata in them).

I haven't tried mp3tag yet, but I went through all my files on audacity and removed the metadata and resampled them to 44100hz, but no files after my jingle play. Just silence. I'll try mp3tag but it's a little worrying.

nomakewan, 2025-03-31:
What is worrying about using mp3tag?

You have not provided enough information in your response for us to be able to help further. Let us perform some sanity checks to fix that.

1. Your SD card should have 423 WAV files numbered 1 through 423. If you have more or less than 423 you have an issue and should start with a fresh card and a fresh copy of the GPStar pack/wand sound file archive. If you have 423 sound files numbered 1 through 423, continue.
2. Any and all custom music/sounds should have their filename start with the number 500 and increase by 1 for each.
3. Any and all custom music/sounds must be Uncompressed WAV format, 16-Bit Signed Stereo PCM at 44100 Hz.

Assuming point 1 is true, please provided the MediaInfo information for your custom file 500, 501, and 502 _as they appear on your SD card right now_. We can go from there.

The-Y33TER, 2025-04-02:
> What is worrying about using mp3tag?
nothing. that's... not what i was saying is worrying. I was saying that the fact these issues are arising when the audio files fit the description of a fixed file.

nomakewan, 2025-04-02:
> nothing. that's... not what i was saying is worrying. I was saying that the fact these issues are arising when the audio files fit the description of a fixed file.

This is actually very common, especially since we have evidence that your files were in the wrong format. I would not worry too much.

Instead, we should follow the scientific method and make small changes and see if we achieve the desired result. Right now, the important thing will be making sure the base sound effects are as they should be, and then adding a small number of music files (500, 501, 502) and assuring that those three files all are in the correct format.

The-Y33TER, 2025-04-02:
> You have not provided enough information in your response for us to be able to help further. Let us perform some sanity checks to fix that.

sorry, I'm being pulled in a bunch of different directions right now with finals season. 

> 1. Your SD card should have 423 WAV files numbered 1 through 423. If you have more or less than 423 you have an issue and should start with a fresh card and a fresh copy of the GPStar pack/wand sound file archive. If you have 423 sound files numbered 1 through 423, continue.

I do. 

> 2. Any and all custom music/sounds should have their filename start with the number 500 and increase by 1 for each.

They are and do. 

> 3. Any and all custom music/sounds must be Uncompressed WAV format, 16-Bit Signed Stereo PCM at 44100 Hz.

I believe they are. I will double check tomorrow afternoon, but I exported them all as such when I went through and resampled. 

> Assuming point 1 is true, please provided the MediaInfo information for your custom file 500, 501, and 502 _as they appear on your SD card right now_. We can go from there.

Will do.

The-Y33TER, 2025-04-02:
> Right now, the important thing will be making sure the base sound effects are as they should be, and then adding a small number of music files (500, 501, 502) and assuring that those three files all are in the correct format.

The base sound effects are formatted correctly and are proper, I reformatted the drive (to be safe) and put the current update's SFX on, then when I put the new music on I pasted the batch of new metadata stripped 44100hz files. The only thing I can think of that is a problem is there's still retained metadata (that I have yet to use mp3tag to look for, haven't been on my pc since sunday) or the other error presented by the mediainfo readout.

nomakewan, 2025-04-21:
Checking in on this. Are you able to get the MediaInfo readouts for music files 500, 501 and 502 from your SD card? Or were you able to resolve this issue independently in the interim?

nomakewan, 2025-04-24:
I'm going to close this for now; if you still need assistance, please feel free to reopen and provide the requested information to continue diagnostics.

The-Y33TER, 2025-05-08:
> I'm going to close this for now; if you still need assistance, please feel free to reopen and provide the requested information to continue diagnostics.

Sorry!! I've been caught up with a bunch of school stuff and haven't found the free time to rip the pack open again (I need to get a usb-c FDTI board and an sd extender, bad.)

The-Y33TER, 2025-05-08:
> Checking in on this. Are you able to get the MediaInfo readouts for music files 500, 501 and 502 from your SD card? Or were you able to resolve this issue independently in the interim?

Yes! It fed me the following:

``General
Complete name                            : D:\500_Jingle.wav
Format                                   : Wave
Format settings                          : PcmWaveformat
File size                                : 1.21 MiB
Duration                                 : 7 s 175 ms
Overall bit rate mode                    : Constant
Overall bit rate                         : 1 411 kb/s

Audio
Format                                   : PCM
Format settings                          : Little / Signed
Codec ID                                 : 1
Duration                                 : 7 s 175 ms
Bit rate mode                            : Constant
Bit rate                                 : 1 411.2 kb/s
Channel(s)                               : 2 channels
Sampling rate                            : 44.1 kHz
Bit depth                                : 16 bits
Stream size                              : 1.21 MiB (100%)

General
Complete name                            : D:\501_Bernstein.wav
Format                                   : Wave
Format settings                          : PcmWaveformat
File size                                : 30.5 MiB
Duration                                 : 3 min 1 s
Overall bit rate mode                    : Constant
Overall bit rate                         : 1 411 kb/s

Audio
Format                                   : PCM
Format settings                          : Little / Signed
Codec ID                                 : 1
Duration                                 : 3 min 1 s
Bit rate mode                            : Constant
Bit rate                                 : 1 411.2 kb/s
Channel(s)                               : 2 channels
Sampling rate                            : 44.1 kHz
Bit depth                                : 16 bits
Stream size                              : 30.5 MiB (100%)

General
Complete name                            : D:\502_WalkTheMoon.wav
Format                                   : Wave
Format settings                          : PcmWaveformat
File size                                : 38.1 MiB
Duration                                 : 3 min 46 s
Overall bit rate mode                    : Constant
Overall bit rate                         : 1 411 kb/s

Audio
Format                                   : PCM
Format settings                          : Little / Signed
Codec ID                                 : 1
Duration                                 : 3 min 46 s
Bit rate mode                            : Constant
Bit rate                                 : 1 411.2 kb/s
Channel(s)                               : 2 channels
Sampling rate                            : 44.1 kHz
Bit depth                                : 16 bits
Stream size                              : 38.1 MiB (100%)``

nomakewan, 2025-05-08:
Those three files look correct. Please do see what happens if only those three files are on the card (after the normal pack audio of course).

Also confirm again that (as of 5.4.4) there are only 423 WAV files not numbered 500~ on the SD card. If there are more than 423 for some reason, as mentioned that will cause issues. So there should be files numbered 1 through 423, then 500, 501, 502.

If you have not yet updated to 5.4.4, I highly recommend doing so. Also make sure when you do the update that you use the latest flash tool as we have had reports of non-working boards after flashing with the old tool.

https://gpstartechnologies.com/pages/support-downloads

Very easy to tell which flash tool you have if you are using Windows. The old tool was less than 1MB in size, and the new one is over 6MB in size.

The-Y33TER, 2025-05-08:
> Those three files look correct. Please do see what happens if only those three files are on the card (after the normal pack audio of course).
> 
I'll test this tomorrow afternoon, thank you!
> Also confirm again that (as of 5.4.4) there are only 423 WAV files not numbered 500~ on the SD card. If there are more than 423 for some reason, as mentioned that will cause issues. So there should be files numbered 1 through 423, then 500, 501, 502.
> 
I will check this.
> If you have not yet updated to 5.4.4, I highly recommend doing so. Also make sure when you do the update that you use the latest flash tool as we have had reports of non-working boards after [...]

I have updated to 5.4.4, works great. I used the old tool, as the new one crashed and refused to open again, so I tested the update w/ the old tool, updated instantly and I must say this update brings some kick ass features. I thank you very much for the independent wand volume, now it operates more akin to the screen shown sfx and the yellow-er vent light is a nice touch.

nomakewan, 2025-05-08:
Looking forward to your follow-up.

Regarding the crash, was there any error message? As far as I know, the only requirement for the Windows version of the flash tool is .NET Framework 4.8. The vast majority of people already have that on their Windows installs, but in case you do not, [here is the redistributable installer from Microsoft](https://go.microsoft.com/fwlink/?linkid=2088631).

The-Y33TER, 2025-05-11:
> Looking forward to your follow-up.
>
> Regarding the crash, was there any error message? As far as I know, the only requirement for the Windows version of the flash tool is .NET Framework 4.8. The vast [...]

It sat attempting to update the wrong port (I accidentally selected port 4 instead of 5) for 10 minutes until i got bored of sitting and waiting for it so I killed it from task manager. Guessing that corrupted it, I’ll redownload it later, for the next update.

nomakewan, 2025-05-11:
> It sat attempting to update the wrong port (I accidentally selected port 4 instead of 5) for 10 minutes until i got bored of sitting and waiting for it so I killed it from task manager. Guessing that corrupted it, I’ll redownload it later, for the next update.

Odd that it didn't time out; I'll have to look into why that is. In my testing if there wasn't a valid Mega 2560 at the other end, it would error out pretty quickly. As for not working, that could be because you killed the parent process but not the child process. That is, you killed the GPStar flasher, but didn't kill the instance of `avrdude` it spawned while running. If `avrdude` is still stuck running in the background, it's not going to allow a new instance of the GPStar Flasher to overwrite it.

Very good feedback, I'll have to see if I can program a way around these sorts of errors.

nomakewan, 2025-05-12:
@The-Y33TER Could you please let me know specifically which serial port hardware you have installed on your computer that caused the hang when you selected its COM port from the drop-down menu? The information about this device from your device manager would be incredibly helpful.

While I was able to reproduce the issue using a null modem emulator and have reported the issue to the maintainers of the programming tool, they are having trouble reproducing the same issue and so I would greatly appreciate any help you can provide in finding out why certain serial ports hang rather than timing out.

The-Y33TER, 2025-05-13:
> [@The-Y33TER](https://github.com/The-Y33TER) Could you please let me know specifically which serial port hardware you have installed on your computer that caused the hang when you selected its COM [...]

The "updated" version directly from the support & downloads page, circa last last weekend (over 7 days ago). Downloaded it, tried to update, failed, killed it, moved on to older flasher. I may have had my pack SD card inserted at the time, and it seems that the software may have identified my laptop's TF slot as a serial capable COM port.

nomakewan, 2025-05-13:
If by any chance you could replicate those conditions (no need to actually do the flash part) and see what shows up in the "COM" port drop-down and then let us know what that COM port belongs to according to the Device Manager that would be really cool.

I mean, it has nothing to do with the music problem, but the folks who made the AVR flash tool are quite curious what hardware can trigger that bug.

The-Y33TER, 2025-05-14:
> If by any chance you could replicate those conditions (no need to actually do the flash part) and see what shows up in the "COM" port drop-down and then let us know what that COM port belongs to [...]

gotcha ill try

The-Y33TER, 2025-05-28:
> Thank you, I look forward to the results.
> 
> Also no need to do the COM port test thing; the AVR flash tool guys found and fixed the problem, and the fixes are already a part of the GPStar flash tools. :) Let's just focus on the music problem and see what we can do.

I saw the COM port thread on your account (i was wondering the other day if you had figured it out with them) and I’m happy to hear the issue has been resolved :> 
I’m gonna test the music stuff in a couple minutes, I just gotta grab my computer. Should I update to 6.1?

nomakewan, 2025-05-28:
> I saw the COM port thread on your account (i was wondering the other day if you had figured it out with them) and I’m happy to hear the issue has been resolved :> I’m gonna test the music stuff in a [...]

6.0.1, yes, go right ahead. 6.1.Dev, absolutely not.

The-Y33TER, 2025-05-29:
> 6.0.1, yes, go right ahead. 6.1.Dev, absolutely not.

I'll update it later. I tested it with just those three files (the ones mediainfo read as totally ok) and no playback for the songs, still successful playback on the jingle.

nomakewan, 2025-05-29:
That is so strange. I just don't get how this is continuing to be an issue.

I hate to have to ask this, but would you be willing to put all three of those files onto a temporary file share? Just add them to a ZIP file and then upload it to a service such as https://litterbox.catbox.moe (with a 12-hour timeout, just in case I'm not at my PC the moment you post).

I'd like to take a look at the files and see what happens when I put them on my own SD cards to see if I can replicate this locally. By all accounts it should be working.

The-Y33TER, 2025-05-29:
3 Files (In order) 
**500_Jingle.wav**
https://litter.catbox.moe/vmfuam.wav
**501_Bernstein.wav**
https://litter.catbox.moe/d5mro0.wav
**502_WalkTheMoon.wav**
https://litter.catbox.moe/5rydje.wav

nomakewan, 2025-05-29:
https://github.com/user-attachments/assets/e88a5451-6ff5-43b6-9a0c-1713a756e3bf

Worked perfectly first try. All I did was delete the music files that were already on my card and copy and paste the ones you uploaded. So once more I'm afraid I don't know what's going on with your setup.

So all I can do is cover the basics.

1. Are you on firmware 5.4.4 or 6.0.1?
2. Do you have a WAV Trigger? If so, make sure it is flashed to firmware 1.40.
3. Do you have a GPStar Audio? If so, make sure it is flashed to firmware 1.04.
4. Try to reformat the microSD card and start with a fresh copy of the SFX and music files.

Speaking of, which microSD card are you using again?

The-Y33TER, 2025-05-29:
> musictest.mp4 
> Worked perfectly first try. All I did was delete the music files that were already on my card and copy and paste the ones you uploaded. So once more I'm afraid I don't know what's going on with your setup.
Odd.
> 1. Are you on firmware 5.4.4 or 6.0.1?
5.4.4. I'll update to 6.0.1
> 3. Do you have a WAV Trigger? If so, make sure it is flashed to firmware 1.40.No.
> 4. Do you have a GPStar Audio? If so, make sure it is flashed to firmware 1.04.
Not sure, I'll flash 1.04.
> 6. Try to reformat the microSD card and start with a fresh copy of the SFX and music files.
Gotcha.
> 7. Speaking of, which microSD card are you using again?
Generic card, same specs as the suggested SanDisk ones, just walmart brand instead. It runs all the other files fine, so I doubt that's the issue.

nomakewan, 2025-05-29:
It's too bad you don't have a WAV Trigger, honestly; I have a test-use firmware for it that lets you test the performance of your SD card to see if it's hiding any nasty surprises. We don't currently have any similar firmware for the GPStar Audio.

The-Y33TER, 2025-06-01:
IT WORKS? 
I turned on the music, jingle played, then out of curiosity i toggled on and off the looping and the bernstein theme started playing. no clue.
Maybe a bug? i might be crazy. The 5th chunk wasn't solid on, meaning no loop, but it refused to play anything after the jingle until i turned loop on and off.

nomakewan, 2025-06-02:
Wait. You mean to say that this whole time, you never actually tried using the music navigation controls? You were just hitting play and then leaving it there?

The-Y33TER, 2025-06-02:
> Wait. You mean to say that this whole time, you never actually tried using the music navigation controls? You were just hitting play and then leaving it there?

No. I had tested the playback controls, I believe there was an issue with the files when I resampled them, which after reloading them I corrected, earlier today. However, the playback does still appear to be broken slightly, the first track plays then doesn’t advance correctly unless i turn on and off the loop while it’s running.

The-Y33TER, 2025-06-02:
> More pertinent question: regardless of loop setting, does the manual next/previous track control work?

Yes, they work fine, now.

nomakewan, 2025-06-02:
Okay. So there is potentially an issue regarding either the initial state of the loop setting, or the initial state of the music check timer. But there is no actual issue regarding reading the music files or indexing of music files.

I'm going to reopen this issue and then take a peek at the code to see what I can find. Thank you for the clarification, it helps immensely.

The-Y33TER, 2025-06-02:
> Ran a test with my pack that only has a wifi module attached (no wand). Hitting play with your files and just leaving it there immediately after power-on works just fine; it proceeds to the next track after the jingle finishes after about 8 seconds.
> 
> Could you confirm that immediately after power-up, if you start playing music and just leave it alone, it does not proceed to the next track automatically after 10 seconds? If it does proceed, how long does it take to do so?

I'm in bed, currently. I'll check tomorrow afternoon. Formerly, the jingle would finish and never advance, I tested it yesterday and it didn't advance after 30 seconds. I have yet to test this after the loop on/off cycle. I'll see.

nomakewan, 2025-06-10:
Hey there, gentle reminder to test out 6.1.Dev and see if it resolves this issue for you.

The-Y33TER, 2025-06-18:
> Hey there, gentle reminder to test out 6.1.Dev and see if it resolves this issue for you.

Sorry, forgot to respond to this. I'll wait until the next branch is public, I have zero clue how to compile the code, even after reading COMPILING-FLASHING and VSCODE. (I'm sorry for being a moron)

The-Y33TER, 2025-06-20:
> Please let me know ASAP whether or not this resolves your reported issue.

It did not.

The-Y33TER, 2025-06-21:
> > It did not.
> 
> I have created a debug Proton Pack firmware. Please keep your wand and wifi/Attenuator on the `6.1.Dev` firmware linked earlier, but flash this to your Proton Pack:
> 
> https://github.com/nomakewan/GPStar-proton-pack/blob/musictest/binaries/pack/ProtonPack.hex?raw=1
> 
> With this firmware installed and music running, you will receive three different voice callouts as the music system runs:
> 
> 1: Check Music timer completed, Music Status Check timer reset (song playing and not yet ended). 2: Check Music timer completed, track end detected, next track queued. 3: Next track timer completed, next track commanded to play.
> 
> If you can take a video showing what's going on while this firmware is installed that would be very helpful.

Can do. Gimme like 15 minutes.

The-Y33TER, 2025-06-21:
> I have created a debug Proton Pack firmware. Please keep your wand and wifi/Attenuator on the `6.1.Dev` firmware linked earlier, but flash this to your Proton Pack: [...]

https://files.catbox.moe/n23mrb.mov

The-Y33TER, 2025-06-22:
> > https://files.catbox.moe/n23mrb.mov
> 
> This appears to be working perfectly correctly as-coded. What appears to be the issue?
 
I don't know, I'm just doing what ya told me. I was under the impression you were trying to fix the big gap in dead air between the end of the jingle and the next track.

nomakewan, 2025-06-22:
No, my assumption from the way you were reporting this originally was that the music was not automatically proceeding _at all_.

The 20.5 second delay is absolutely expected, and in fact I mentioned it earlier:

>Second, much of what you found is actually because the jingle is so short, at 7175 milliseconds. When you first start playing music, ms_music_status_check is started with a timeout of 20000 milliseconds (20 seconds). So in fact it would take another 12.825 seconds to trigger the status check. Assuming this status check is triggered correctly, it would then wait an additional 500 milliseconds before proceeding to the next track, for a total delay between the end of the jingle and the start of the next track of 13.325 seconds.

Shortening this initial delay could potentially cause issues (it increases serial bus activity) so I'll discuss it with the other devs to see how much shorter we can make that without negatively impacting performance. But for now, since it appears the actual bug is resolved, I'm going to go ahead and close this issue.

closed 2025-06-22 by nomakewan