gpstar81/GPStar-proton-pack #429: [Bug]: Sputtering audio on pack

adammac79, 2025-04-28:
### What actions were attempted?

Reporting bug via request from Michael: Pack audio exhibits very short sputtering dropouts during polyphonic wav playback of induced sounds like firing the wand. It sounds a bit like clipping or the speakers are being overdriven, however It occurs at all volumes. It's very noticeable during firing when it sounds like overdriving, quieter/cleaner sequences like post firing ramp-down sounds when it sounds like quick cuts. The sputtering occurs in random spots along the playback period, and not exhibited on the wand audio. Confirmed by swapping audio cards between pack and wand and listening with headphones through the aux ports. 

### What actually happened?

Michael advises he believes it is due to different amplification levels in the code, so he asked me to post about it here so a dev could look at it.

### What was expected?

Cleaner sound was expected

### Firmware Version

5.4.4

### Mode in Use

Any / Both Modes

### Notable Hardware

GPStar: pack and wand boards, audio cards, amp, recommended speaker set. DIY wifi module ("attenuator"), DIY vent smoke. 

### Homework Completed

- [x] I did my part!

adammac79, 2025-04-28:
Yes, I updated both audio boards today along with 5.4.4.

nomakewan, 2025-04-28:
I did check the code just to be sure, and the gain control on the packside was not altered in 5.4.4. It still specifies that the maximum possible gain on the pack is +0, which is unity gain:

https://github.com/gpstar81/GPStar-proton-pack/blob/main/source/ProtonPack/Audio.h#L47

This means the packside is not allowed to overdrive the output as the wand can (because the wand has a speaker amplifier built in and so can go above unity gain). Your wand works without distortion, so clearly this is not the issue. On top of that, the "volume" control on the pack actually adjusts the master gain value and so lowering the pack volume should get rid of the distortion if the master gain is a contributing factor.

As a test, could you flash your pack and wand back to 5.4.3 and re-check?

adammac79, 2025-04-28:
Even during testing at 50% gain the effect is still heard. It's as if there is a struggle to process the additional layer/s of audio on top of the bed audio. To be clear, the problem existed prior to the 5.4.4 update.

nomakewan, 2025-04-28:
Oh, it existed before 5.4.4? That changes everything, I thought this was a brand new report for 5.4.4. Okay, phew.

Glitching, clipping audio when layering the polyphonic sounds is a known issue with SD cards that have sub-par random read performance. I would highly suggest trying a different SD card in your pack board.

I've had success with SanDisk Extreme and Samsung EVO Plus cards in my WAV Trigger and GPStar Audio boards. The important spec for our use case is random read, which unfortunately is not a spec that manufacturers readily provide (they provide sequential burst read because it's always a big fancy number, but this doesn't mean jack diddly for our use case).

You could also try performing a full reformat of the SD card using the [gui32format tool](https://github.com/gpstar81/GPStar-proton-pack/raw/refs/heads/main/extras/sound/guiformat.exe?raw=1) in case the issue is being exacerbated by a fragmented filesystem on the SD card.

adammac79, 2025-04-28:
OK, so I was using SanDisk Extreme Plus, which contrary to what any sane person might assume, is of a lower spec than the "non-Plus" Extreme variant. I have ordered replacements!

nomakewan, 2025-04-28:
> OK, so I was using SanDisk Extreme Plus, which contrary to what any sane person might assume, is of a lower spec than the "non-Plus" Extreme variant. I have ordered replacements!

That'll do it! The Extreme Plus has higher write speeds than the Extreme, but much worse read performance. This works great in something like a video camera or dashcam, but is not a good match for something like a polyphonic audio player.

Let us know how it goes!

nomakewan, 2025-05-01:
Issue was confirmed to be poor quality SD cards, and that a SanDisk Extreme SD card solved the problem.

closed 2025-05-01 by nomakewan

adammac79, 2025-05-01:
I think it might be helpful to add that what solved it in the end was moving from the 32gb extreme cards to the 64gb ones. Reason being is that the 32gb cards only read up to 100mb/s where 64 and up go to 170mb/s. You may want to rev your documentation. I will fully admit that the issue may not be noticed by everyone, but I am really focused on sound.

nomakewan, 2025-05-01:
I have reason to believe your non-working microSD cards were counterfeit. I saw the performance chart you provided as part of the diagnostics and those numbers did not match what would have been expected from any card in SanDisk's Extreme lineup.

For comparison, here is a legitimate 32GB SanDisk Extreme Plus microSD card, purchased from a brick & mortar electronics retailer in the USA:
[image]
[image]

Compare this to the 32GB card you tested, which showed 20 MB/s read and 36 MB/s write. Especially for a UHS Class 3 card (which the SanDisk Extreme, Extreme Plus, and Extreme Pro cards all are) one would expect significantly higher sequential read than sequential write.

For additional benchmarking, here is a 64GB SanDisk Extreme Plus:
[image]
[image]

And a Samsung Pro Enduranec 32GB:
[image]
[image]

Notice the brief dip in read performance on the 64GB card; this is an artifact of using GUIFormat to format the 64GB card as FAT32, since Windows will not allow you to do that natively. Unfortunately for whatever reason, using this tool to format a card as FAT32 always has a detrimental effect on read performance near the start of the card. This is why cards 32GB and smaller (if you can find them) are still recommended for use with the WAV Trigger and GPStar Audio.

That being said, I do agree with the need to update the documentation, and should add ways of testing SD card performance to said documentation as well. Thank you for the feedback, and glad to know the problem is sorted!