gpstar81/GPStar-proton-pack #435: [Feature}: Sounds for firing attempt with Safety on

nomakewan, 2025-05-07:
### Discussed in https://github.com/gpstar81/GPStar-proton-pack/discussions/434

<div type='discussions-op-text'>

<sup>Originally posted by **Toy203** May  7, 2025</sup>
During day use (lol) i found it's a bit weird pressing the intensify button with the barrel retracted and not having any feedback. Would be cool  if the pack played a sound for the "misfiring" action (somehow i'm thinking at that Millenium Falcon hyperdrive failing sound) signalling the user they're trying to shoot with safety on. Could even make the wand vibrate in an erratic pattern, like the pack is barely holding the power in.</div>

ThePackDoctor, 2025-07-27:
Maybe a rebuilt version of the firing failure when Trevor attempts to fire the gunner seat pack and fails due to minipuft damage?

nomakewan, 2025-10-03:
This will be included alongside the 6.1.0 release!

closed 2025-10-03 by nomakewan