gpstar81/GPStar-proton-pack #446: [Feature]: Better Mode Configuration Options.

The-Y33TER, 2025-06-04:
### What would you like to add in terms of software changes?

I'm sitting here, loving the custom and TVG modes, but I'm realizing, why isn't it an option to have the ability to turn off specific spectral modes? I don't use the rainbow nor holiday and I wish I could turn them off without having to flash my wand with custom firmware (obviously the issue is less that I'd have to set the config.h values to false, but it's that I'd have to do it for _every single_ update. 

Also, an option to change the default mode (persisting across battery cycles) would be pretty neat. A buddy of mine is gonna get the standalone wand kit and we built his pack to be Meson themed, so having him start out Meson would be pretty nice.
ALSO! Having separate sfx for the custom mode would be convenient. I'm gonna set mine up to be my "overclocked" mode, and I want to add more depth to my stream sfx (hence, the overclocked nature of the mode)

### Would this request involve any specific hardware?

Nope.

### Homework Completed

- [x] I did my part!

DustinGrau, 2025-06-09:
So technically that does exist, it's not just really bubbled up to the user level. Yet.

The-Y33TER, 2025-06-09:
> So technically that does exist, it's not just really bubbled up to the user level. Yet.

It would be much appreciated if it could bubble up to the user level at some point during this branch (if possible) :D

ThePackDoctor, 2025-12-29:
> I've just realized this request existed, and I happened to have been refactoring a lot of the stream selection logic already, so I'm looking into adding individual flags to enable/disable the VG modes (Stasis, Slime, Meson) rather than just a generalized "VG" stream-enablement flag.

Even better than I was hoping for. Sick!