gpstar81/GPStar-proton-pack #453: [Feature]: Choose slime colour independent of year

yourevilclone, 2025-07-25:
### What would you like to add in terms of software changes?

It would be great to be able to choose pink slime while in Afterlife mode, rather than needing to switch to 1989 mode. (Maybe this is already planned for the custom modes of version 6.)

In fact, it would be great to have fully customisable colours.

I have a videogame-style mini Slime Blower attached to my wand, that shoots silly string. But here in the UK there is only one brand of silly string can fits it, and that brand can only be bought in bulk online, so as well as pink and green I end up with a bunch of other colours. Red and purple are close enough to pink, but I've not got much use for the yellow and blue. But with customisable slime colours, can use them up too!

### Would this request involve any specific hardware?

No

### Homework Completed

- [x] I did my part!

Toy203, 2025-07-25:
Yeah I'm going to need to some some pics of this awesome contraption

nomakewan, 2025-07-25:
This is already planned for the "custom template" feature in v6.

You will not be able to override Afterlife itself; each year mode will be completely fixed feature-wise. 1984, 1989, 1991, 2021, 2024.

However, there will be a 6th slot for "Custom" where you can mish-mash any feature you like into your own personal template. Bargraph animations, sound effects, cyclotron effects, power cell delays, color schemes, everything. Make it what you want.